stephane/libmodbus #595: Could we replace the strlcpy(...) function with strncpy(...) function?

huskier, 2021-08-05:
## libmodbus version

  <3.1.6>

## OS and/or distribution

  <Windows 10, QT 5.15.2, mingw810_64>

## Environment

  <CPU architecture 64 bit>

## Description

When we use the strlcpy(...) function to compile, there is an error occuring, which reports undefined reference to strlcpy(...) function. We searched the QT environment, there is no BSD's string.h there, and only an official string.h, but strlcpy(...) is not there. Could we replace the strlcpy(...) function with strncpy(...) function?

jan-kardell, 2021-08-29:
In modbus.c there is a definition of strlcpy() for systems that lack it. Maybe something is not configured correctly in your build?

The strncpy() function is not equivalent, as it do not guarantee that the result is NULL terminated.

huskier, 2021-09-01:
My fault, I have to comment out the line of HAVE_STRLCPY, and it is not OK to define it zero.
/* Define to 1 if you have the `strlcpy' function. */
#define HAVE_STRLCPY 0

Thanks.

closed 2021-09-01 by huskier