stephane/libmodbus #605: Can anyone give me a ready2use win32 libmodbus.dll?

ottelo9, 2021-11-08:
Hi there. Can anyone give me a compiled dll ? Or is there a download url somewhere?

Cloud-lee, 2021-11-10:
Try this

ottelo9, 2021-11-10:
Ok I compiled it by myself with Visual Studio Express C++ in a virtual machine. Here are the .dll .lib and header files:

[https://drive.google.com/file/d/1uziMsiYTOsz_k3_wNsMiDr3y1vjwqwmV/view?usp=sharing](url) libmodbus_win32_v3.1.6.zip

stephane, 2022-01-08:
Could you store this information in the wiki and better, provide of proper public that I can publish on libmodbus.org?

ashishsiddharth, 2022-01-17:
I couldn't run the solution file using visual studio 2022, got several errors. I don't know why. But  Thank you for providing ready 2 use DLL files.

ottelo9, 2022-02-21:
> Could you store this information in the wiki and better, provide of proper public that I can publish on libmodbus.org?

https://github.com/stephane/libmodbus/wiki/Compile-dll-under-Windows-with-Visual-Studio-2008-Express

What do you mean "provide of proper public" ?

stephane, 2022-02-24:
I meant an URL not hosted on Google Drive.
Thank you for the wiki.

Feel free to close the issue now.

ottelo9, 2022-02-25:
> I meant an URL not hosted on Google Drive. Thank you for the wiki.
> 
> Feel free to close the issue now.

How can I add it to the release page (assets?)

ottelo9, 2022-04-25:
Download Link: https://www.sendspace.com/file/cb850k

closed 2022-04-25 by ottelo9