stephane/libmodbus #608: The set response timeout is 200ms, but the actual time is 400ms

teaczjf, 2021-11-26:
hi！
i set the timeout
                      
  modbus_set_response_timeout(pObj->m_modbus, 0,200* 1000); //timeout=200ms

but the result turn out is 400ms

this is my code

LOG(WARNING) << "#######################开始读取"<< "modbusid_cur= " << modbusid_cur;
ret = modbus_read_registers(pObj->m_modbus, item->second.address(), 8, checkbuff);
LOG(WARNING) << "#######################结束读取" << "modbusid_cur= " << modbusid_cur ;
this is my log
2021-11-26 10:39:45,317 #######################开始读取modbusid_cur= 2 slave_id_cur= 2
2021-11-26 10:39:47,719 #######################结束读取modbusid_cur= 2 slave_id_cur= 2

Jasper-Bai, 2022-11-09:
same problem for me. have u solved it ? could u tell me?

teaczjf, 2022-11-09:
收到
      以上为自动回复