stephane/libmodbus #615: Issues with initializing the registers in the server

ram469, 2021-12-22:
Hi,

I am using the below statement to create a memory with different start addresses. 
```
modbus_mapping_t *mb_memory_section;
mb_memory_section = modbus_mapping_new_start_address(start_bits, nb_bits, start_input_bits, nb_input_bits, start_registers, nb_registers, start_input_registers, nb_input_registers);
```
then I tried to initialize this memory like below
`mb_memory_section ->tab_registers[ start_registers ] = 555;
`
I tried to get the value using the client at  start_registers  address but it is zero.

Worked case:
When I initialized the memory like below it is working
`mb_memory_section ->tab_registers[0] = 555;
`
I tried to get the value using the client at this `start_registers `address and it is 555

Is this the expected behavior? or a bug.

Hardware: NVIDIA Jetson Xavier NX board
OS: Ubuntu
libmodbus version: latest (built from sources)

Thanks,
Ramakrishna

stephane, 2022-01-08:
What value do you use for `start_registers`?
How many registers did you allocate?

ram469, 2022-01-18:
Hi Stephane,

Sorry for the late reply, I use any random value rather than 0 and <= 65535. I have allocated 10 registers.

Thanks,
Ramakrishna

stephane, 2022-08-17:
The next documentation will include a better explanation:

The different starting addresses make it possible to place the mapping at any
address in each address space. This way, you can give access to the clients at
values stored at high addresses without allocating memory from the address zero,
for eg. to make available registers from 340 to 349, you can use:

```c
mb_mapping = modbus_mapping_new_start_address(0, 0, 0, 0, 340, 10, 0, 0);
```

The newly created `mb_mapping` will have the following arrays:

- `tab_bits` set to NULL
- `tab_input_bits` set to NULL
- `tab_input_registers` allocated to store 10 registers (`uint16_t`)
- `tab_registers` set to NULL.

The clients can read the first register by using the address 340 in its request.
On the server side, you should use the first index of the array to set the value
at this client address:

```c
mb_mapping->tab_registers[0] = 42;
```

closed 2022-08-17 by stephane

PlayaJay22, 2023-11-01:
I think your documentation has an error. parameter 5 and 6 are for holding registers, not input registers.
I think your documentation on libmodbus.org should be:

_The newly created mb_mapping will have the following arrays:

tab_bits set to NULL
tab_input_bits set to NULL
tab_registers allocated to store 10 registers (uint16_t)
tab_input_registers set to NULL._

Unless I'm missing something fundamental, but I've looked into your modbus.c code and it clearly shows parameters 5 and 6 are for nb_registers, NOT nb_input_registers

Please update the documentation because this is SUPER confusing.