stephane/libmodbus #621: modbus_new_tcp_pi allocates way more memory than modbus_new_tcp and has arbitrary length limits on its string arguments

psychon, 2022-01-09:
Hi!

## libmodbus version

Sure, but I am not using any version:
```
$ pkg-config --modversion libmodbus
Package libmodbus was not found in the pkg-config search path.
Perhaps you should add the directory containing `libmodbus.pc'
to the PKG_CONFIG_PATH environment variable
Package 'libmodbus', required by 'virtual:world', not found
```

## OS and/or distribution

```
$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux bookworm/sid"
NAME="Debian GNU/Linux"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

## Environment

Since you apparently want to know my architecture, here's my CPU:

Intel(R) Core(TM) i7-6700K CPU @ 4.00GHz

## Description

The "protocol independent" TCP code uses this struct:

https://github.com/stephane/libmodbus/blob/3da2d01916ef118aba30a1951d89e0ca0f90e2f7/src/modbus-tcp-private.h#L30-L42

This has arbitrary length limits of 1024 bytes of the node length and 31 bytes for the service.

Also, this allocates more than 1 KiB of memory even when I just provide 10 characters. That is 99% wasted memory for storing two short strings.

The implementation returns an error if these limits are reached:

https://github.com/stephane/libmodbus/blob/3da2d01916ef118aba30a1951d89e0ca0f90e2f7/src/modbus-tcp.c#L896-L897

https://github.com/stephane/libmodbus/blob/3da2d01916ef118aba30a1951d89e0ca0f90e2f7/src/modbus-tcp.c#L919-L920

## Expected behaviour

Less memory usage and no arbitrary limits.

## Actual behaviour

This section doesn't really fit this report, so I will instead use this for suggestions.

I see two ways for this:

- Instead of using a fixed size array, do an extra memory allocation. E.g. (change is completely untested and should only show the idea):

```diff
diff --git a/src/modbus-tcp-private.h b/src/modbus-tcp-private.h
index 164c8c7..adef8a9 100644
--- a/src/modbus-tcp-private.h
+++ b/src/modbus-tcp-private.h
@@ -27,18 +27,15 @@ typedef struct _modbus_tcp {
     char ip[16];
 } modbus_tcp_t;
 
-#define _MODBUS_TCP_PI_NODE_LENGTH    1025
-#define _MODBUS_TCP_PI_SERVICE_LENGTH   32
[... 257 lines cut ...]
-        return NULL;
-    }
+    strcpy(ctx_tcp_pi->node_and_service, node);
+    strcpy(&ctx_tcp_pi->node_and_service[node_size + 1], service);
 
     ctx_tcp_pi->t_id = 0;
 
```

## Steps to reproduce the behavior (commands or source code)

Source code is linked above, I guess. I don't have any own code using libmodbus, so....?

## libmodbus output with debug mode enabled

Uhm... `command not found`, I guess? See `pkg-config` output from above.

## Copyright

If you do not worry about copyright issues with my untested sketch patches above, then just ignore this section.

In case you worry: Pick one of the following:

- If you are living in a jurisdiction were this works (I do not): I hereby release all information in this post to the public domain.
- I hereby license all information in this post under the WTFPLv2. See https://en.wikipedia.org/wiki/WTFPL. I explicitly mention that I think this allows to act as if you own all copyright: You can rewrite, change the license, sell, and whatever else you want.

stephane, 2022-01-25:
LGTM.
Could you provide a PR, please?

stephane, 2022-08-17:
I choose the proposal with 2 mallocs, way easier to understand and maintain ;)

commit ef3c4bc9 (2022-08-17): Reduce memory use of TCP PI backend (closes #621)

- allocate exact memory required to store node and service strings
  instead of around 1kb of static memory.
- accept NULL value of service to use default Modbus port number (502)
- unit test updated

The new documentation will be updated in another commit.
closed 2022-08-17 by stephane