stephane/libmodbus #625: How to write into signle holding register

Adambrin, 2022-01-18:
Hello, i installed the library, i tested coils, bits succesfully but i dont know how to test write siggle holding register. Say i want to write the value 260 to holding register address 32. Can you please show i can put these to examole unit test client?

mhei, 2022-01-18:
This depends on what your device supports: [modbus_write_register(3)](https://libmodbus.org/docs/v3.1.6/modbus_write_register.html) or [modbus_write_registers(3)](https://libmodbus.org/docs/v3.1.6/modbus_write_registers.html)

Adambrin, 2022-01-18:
As a matyer of fact i want to test  both of them (single holding  register  and many holding  registers)

Adambrin, 2022-01-19:
Here's the output when i try to write single register:

Opening /dev/ttyS2 at 38400 bauds (N, 8, 1)

** UNIT TESTING **

1/1 No response timeout modification on connect: OK

[01][06][00][21][01][4D][18][65]

Waiting for a confirmation...
ERROR Connection timed out: select
Bytes flushed (0)
1/2 modbus_write_register: 
Line 212: assertion error for 'rc == 1':

Can you advice?

Adambrin, 2022-01-19:
Problem solved. Thanks

mhei, 2022-01-19:
Then you should close this issue.

closed 2022-02-21 by stephane