stephane/libmodbus #638: Is there a modbus TCP with TLS ?

schnitzlein, 2022-06-03:
Hi,

is there currently some ongoing task to implement the modbus TCP with TLS Support? as documented in libmodbus org?

Can I find it some where? would be really cool!

greetings

schnitzlein, 2022-06-15:
Maybe I am overseeing something...
But it would be cool if I can use libmodbus with TLS support, because we live in 2022 and Encryption is mostly mandatory.

@stephane do you know maybe any project which currently supports modbus TCP + TLS ?
Any kind of information would be helpful :) 
(commercial lib ... secret at ... or any open source lib containing it)

My research results was only that I was able to find a port for Arduino OS seems like they have it for ESP8266, which is quite specific. I am looking more for a generic solution for Linux.

Or ... write it by myself, if there is none solution yet.

stephane, 2022-09-27:
No but it could be a good fit for promodbus spin-off before merging it later in libmodbus...

alongL, 2022-12-28:
@schnitzlein  Before it is ok, maybe you can use some ssl tunnel software to solve this problem.
for example:  https://github.com/ghostunnel/ghostunnel   etc.

seboste, 2024-01-04:
@stephane Is there any ongoing/planned work regarding this in the promodbus spin-off as you insinuated earlier? Thanks.

LeSpocky, 2025-12-09:
There is PR #560 …