stephane/libmodbus #645: Simple code compilation problem - linker error LNK2019 - only with RTU

aLAN-LDZ, 2022-07-07:
Hello, when I try to compile my code based on modbus-rtu I get an error with every function from your library

example:

`#include <modbus.h>

int main()
{

    modbus_t* ctx;
    ctx = modbus_new_rtu("\\\\.\\COM5", 38400, 'N', 8, 1);
}`

error:

LNK2019	unresolved external symbol _modbus_new_rtu referenced in function _main.

In case of using functions from the TCP library there is no this problem, unfortunately my device does not support TCP :( Please help if possible

-libmodbus version: 3.1.7
-Windows / Visual Studio 2019
- CPU architecture, 64 bit

SnorlaxGengar, 2022-08-12:
FYI: is your project add the lib path?

aLAN-LDZ, 2022-08-12:
I fixed this problem using a changed library by one of users

mthongkerd, 2022-08-16:
This is what helped me (mscv 2019, x64) so I'm just gonna put it here. It's from this [post](https://groups.google.com/g/libmodbus/c/-eii4OfYcZ8/m/K2ADumMcCAAJ) by Matthew Wachter.

Ok, got it to compile in x64 without too much difficulty. Just set your platform to x64 and make sure your set these settings in the project property pages:

General / Project Defaults / Configuration Type = Dynamic Library (.dll)
Linker / Input / Additional Dependencies = ws2_32.lib

stephane, 2022-08-17:
OK, notes added to next version of the documentation.

closed 2022-08-17 by stephane