stephane/libmodbus #654: How to read unsigned int 32 bit

LucaDavolio, 2022-09-06:
Hi, 
I tried to read a 32 bit word and, as I imagined, it gave me a wrong value.
Is there a way to read 32 bit unsigned integers?
Thanks.

jfsimon1981, 2022-10-10:
Hi,

Check how the device manufacturer implemented it, and reconstruct the 32bit word at your application level from the UINT16 which i believe is the highest bits Modbus supports.

Manufacturer will define how they send data such as signed values, or others. Though for your case you stick the two parts together and reinterpret as 32bit.

Code snippet just to give you an idea

uint32_t data_out;
data_out = (((uint32_t)register_buffer_32b[i] << 16) | uint32_t)register_buffer_32b[i+1]));

modem-man-gmx, 2022-11-29:
my very simplified way of reading 32bit is:

uint16_t Buffer[2];
modbus_read_registers( mbctx, RegAddress, 2, &Buffer[0] );
uint32_t i32Value = MODBUS_GET_INT32_FROM_INT16( Buffer, 0 );

I would not do the bit fiddling by hand, it's already in the libmodbus, have 3.1.6, but am sure it's since longer time there.

I'm waiting a bit for modbus_get_uint32_abcd(), modbus_get_uint32_dcba(), modbus_get_uint32_cdab(), and the like, similar to float. But to be honest, I never got my hands on a real device which needs the MSB/LSB flipping.