stephane/libmodbus #659: What is the status of libmodbus-ascii effort?

jimklimov, 2022-10-12:
Hello, we have had a new NUT driver proposal at https://github.com/networkupstools/nut/pull/1671 which included some combination of libmodbus-ascii sources as needed for that new driver to communicate with an UPS (I am double-checking with the author if that can work with released RTU support instead).

It seems that the libmodbus-ascii sources are around for a quite a while, so there may have been good reasons to not merge them into main codebase and distribute everywhere, e.g. not well enough tested, not too portable, known unreliable, or something around that?

Then there is also a legaleze question about mixing your LGPL with our GPL, if some sources are snatched from the libmodbus project to provide a (fallback at least) implementation for one of many optional programs that can be built in NUT...

Should we be wary about the codebase on technical grounds - that even authors deemed it not mature enough to release, and so insist on rewrite of the submitted NUT driver?

Alternately, maybe it just "fell through the cracks" and you can just merge and release it officially, so NUT with the new driver may just depend on the OS/distro provided builds of libmodbus package (like it already does with some other drivers added in the past year or two)? :)

jimklimov, 2023-10-21:
Um... bump? :)

Over at NUT project we actually have two separate efforts by now that require better libmodbus features than available in the published library (ascii, and now also usb investigated per https://github.com/networkupstools/nut/pull/2063). I would be not too comfortable having NUT publish an alternative fork of the better-featured common library - it would be some burden for us and mostly confusing for end-users (especially if they end up installing two forks and have to resolve conflicts at compile- or run-times), but fear it may come to this for practical reasons...

biergaizi, 2023-10-21:
Unfortunately, in my opinion it appears that active development of the project appears to have nearly ended and now it's in maintenance mode, only fixes and cleanups are made without adding any new feature anymore. The maintenance also seems to be not very active, since not even a [buffer overflow bug reports](https://github.com/stephane/libmodbus/issues/683) reported early this year has received responded or fixed by project maintainers. The only chance that one can bring any real changes in is sending a Pull Request - the good old "patch welcome" - which is fair enough, but most PRs I see are simple fixes. I'm not sure whether more complex patches would be reviewed...

Are the project maintainers still willing to review big patches to bring in major features like libmodbus-ascii? If so, perhaps we can make a upstream attempt. If not, I'm afraid that a fork is inevitable...

stephane, 2024-07-17:
Feel free to rebase the ASCII branch and update it.

closed 2024-07-17 by stephane