stephane/libmodbus #665: modbus_get/set_float_xxxx() mixing bytes

bkggkt3d, 2022-11-03:
After a lot of compile runs for different version I'm now quite sure there is a change in implementation of modbus_get_float_abcd() and modbus_set_float_abcd() funtions, probably all 8 variants are affected.

## libmodbus version

  branch v3.1.8 sources from github 

## OS and/or distribution

  Linux Kernel 3.10 with Debian 11

## Environment

   Linux Debian 10 crosscompile for embedded arm with ./configure --host=arm-linux-gnueabihf

## Description

- modbus_get_float_abcd() from v3.1.8 acts like modbus_get_float_dcba() from v3.1.7, so all 4 bytes need to have different order to get the right float back
- modbus_set_float_abcd() from v3.1.8 acts like modbus_set_float_cdab() from v3.1.7, so the both 16-bit registers came back swapped from the float. Yes, this is even inconsistent to the problem at get-funtion.

## Expected behavior or suggestion

 Up to version 3.1.7 all 8 functions are working as expected and how I understood the documentation

modem-man-gmx, 2022-11-29:
Could you please give some more easy reproducible example and more details of the counterpart device?
F.i. can this be reproduced with widely available hardware like Raspberry Pi? Which distro is to be installed to reproduce?
Finally, you had in mind, that float implementation between hardware (x86/arm) and platforms and compilers may be incompatible? I ask, because mixing IEEE 754, intel-x86-80bit and some 96 bit on a µC (gcc, IIRC) made me some headache in the past.

Would like to verify your observation on my hardware zoo ...

morgoth6, 2023-03-14:
I think same problem hit my project. It's quite easy to reproduce I guess. Not sure I understand this correctly, but run a float by modbus_set_float_badc() && modbus_get_float_badc() should give same float value used at the beginning ?
```
        float f_in = 123456.0f;
        float f_out;

        uint16_t f16[2];
        modbus_set_float_badc(f_in, f16);
        f_out = modbus_get_float_badc(f16);

        printf("in[%f] = 0x%02X%02X%02X%02X\n", f_in, ((uint8_t *)&f_in)[0], ((uint8_t *)&f_in)[1],
               ((uint8_t *)&f_in)[2], ((uint8_t *)&f_in)[3]);
        printf("out[%f] = 0x%02X%02X%02X%02X\n", f_out, ((uint8_t *)&f_out)[0], ((uint8_t *)&f_out)[1],
               ((uint8_t *)&f_out)[2], ((uint8_t *)&f_out)[3]);

```
Gives in result:
```
in[123456.000000] = 0x0020F147
out[-985402689122801928078052294656.000000] = 0x200047F1
```

oe5hpm, 2023-03-14:
the current implementation is obviously wrong.

i did a short bisect on this. following commit breaks things:

49af73debd756be68497a61bd53a07c02673da96 is the first bad commit
commit 49af73debd756be68497a61bd53a07c02673da96
Author: SZ Lin (林上智) <szlin@debian.org>
Date:   Thu Dec 20 13:35:31 2018 +0800

    Fix float endianness issue on big endian architecture.
    
    It converts float values depending on what order they come in.
    
    This patch was modified from rm5248 [1]
    
    [1] https://github.com/synexxus/libmodbus/commit/a511768e7fe7ec52d7bae1d9ae04e33f87a59627

 src/modbus-data.c        | 110 ++++++++++++++++++++++++++++++++++++++---------
 tests/unit-test-client.c |  22 ++++++----
 tests/unit-test.h.in     |  41 ++++++++++++++++--
 3 files changed, 141 insertions(+), 32 deletions(-)

oe5hpm, 2023-03-14:
in detail it looks to me, that the modbus_set_float functions are not correct

ghorwin, 2023-04-05:
Hi,

I can assert that the current implementation is wrong. I stumbled across this when checking conversion from/to cdab float format.

Simple test code:

```c++

float val = 24; 
// on x86 machines in little-endian format this is in memory  "00 00 c0 41"

// 4 byte memory array with two registers/words, each uint16_t also stored in little endian format
uint16_t reg[2]; 
modbus_set_float_cdab( val, reg );

// we expect memory of reg to hold "00 00 41 c0" and this is the case, so the implementation of the
// setter function is correct

// backwards
float val2 = modbus_get_float_cdab( reg );

// memory of val2 is now still "00 00 41 c0", in fact ab and cd get got swapped twice
```

Explanation: consider the function implementations

```
float modbus_get_float_cdab(const uint16_t *src) {
	// we work directly with uint16_t, which are little-endian format as well
	float f;
	uint32_t i;
	uint8_t a, b, c, d;

	a = (src[0] >> 8) & 0xFF;   // here is the bug, as the higher byte of src[0] is not a, but b!
	b = (src[0] >> 0) & 0xFF;
	c = (src[1] >> 8) & 0xFF;
	d = (src[1] >> 0) & 0xFF;
	...
}

void modbus_set_float_cdab(float f, uint16_t *dest)
{
	uint32_t i;
	uint8_t *out = (uint8_t *) dest; // AAH, we work directly on bytes, much better!
	uint8_t a, b, c, d;
	...
}
```

The core problem is the usage of uint16_t and the assumption made when using the shift operation.

The fix is simple, and can be combined with a general code cleanup in this file. The old, outdated swap16 and swap32 macros can go, as the useless memcpy calls.

I'll submit a patch/pull request to fix the issue, shortly.

ghorwin, 2023-04-06:
Hi, just noticed, I was working with wrong assumption, so my fix needs to be revised first.

Consider the transfer from modbus buffer to 16-bit registers: (modbus.c:1311)

```
		for (i = 0; i < rc; i++) {
			/* shift reg hi_byte to temp OR with lo_byte */
			dest[i] = (rsp[offset + 2 + (i << 1)] << 8) | rsp[offset + 3 + (i << 1)];
		}
```

Here, the little/big endianess of the host architecture is already considered. So we have the byte order in memory already matching in each 16-bit register. I'll revise my patch and submit again.

ghorwin, 2023-04-06:
Hi,

here's a variant. What do you think?

```c++
/* Get a float from 4 bytes (Modbus) without any conversion (ABCD) */
float modbus_get_float_abcd(const uint16_t *src)
{
	/* Suppose the modbus byte stream contained the 32-bit float 0x10203040  - abcd.
	   On big endian systems, the memory starting at src contains two 16-bit registers 0x1020  and 0x3040
	   On little endian system, the 16-bit registers memory holds 0x2010 and 0x4030

	   To convert the data to float32 on big-endian we only need to cast the pointer and we are done.
	   On little endian systems, we need to swap the bytes in each word again and then assemble
	   an integer via shift operations and finally cast to float32.

	   A portable way is to retrieve low and high bytes of both words using shift operations, then assemble
	   the 32-bit integer.
	*/

	float f;
	uint8_t a, b, c, d;

	// access buffer memory byte-wise ABCD
	a = src[0] >> 8;     // high byte if first word
	b = src[0] & 0x00ff; // low byte if first word
	c = src[1] >> 8;     // high byte if second word
	d = src[1] & 0x00ff; // low byte if second word

	// assemble in memory location of float
	// from right to left: get address of float, interpret as address to uint32, dereference and write uint32
	*(uint32_t *)&f = (a << 24) | (b << 16) | (c << 8) | (d << 0);

	return f;
}
```

ghorwin, 2023-04-16:
Did anyone check my patch, yet? I'd be happy to get some feedback...

morgoth6, 2023-04-16:
I will check it in next few days.

ghorwin, 2023-05-01:
Hi again, any progress on the issue? I'd appreciate if we introduce a solution to this bug into the current libmodbus, before it gets included in any linux distro packages and breaks existing tools.

Any way I could help?

modem-man-gmx, 2023-05-01:
> we have to address _two_ memory layouts:
> 
> * the source memory layout of the float passed to the conversion function (little-endian/big-endian)
> * the target memory layout expected in the modbus stream (which is part of the modbus protocol specs of a certain device)

Hmmm, I can't say, If I got you right. The "Modubus Application Protocol V1.1b" specification says in section 4.2 (cite:) 

```MODBUS  uses  a  ‘big-Endian’  representation  for  addresses  and  data  items.  This  means that  when  a  numerical  quantity  larger  than  a  single  byte  is  transmitted,  the  most significant byte is sent first. So for example 
  Register size      value 
  16 - bits          0x1234    the first byte sent is   0x12  then 0x34
```

stephane, 2024-07-17:
My take is libmodbus should support non-standard Modbus implementations because there are sadly useful.

closed 2024-07-17 by stephane

ghorwin, 2024-09-18:
See further explanation on #757, as @oe5hpm correctly stated, this is indeed a regression bug introduced in commit 49af73debd756be68497a61bd53a07c02673da96.

Easiest would be to revert this commit.

PR #694 (2024-07-17): Fixes float encoding/decoding for both big and little endian (fixes #665).
Fixes float encoding/decoding for both big and little endian (fixes #665).
Also removes useless memcpy calls and no longer used swap32 and swap16 macros.

This pullrequest modifies a single file. As explained in issue #665, the byte positions in target modbus buffer must be explicitely set according to the modbus protocol specs.

Both for encoding/decoding a cross-platform way working for both big and little endian systems is the use of 32-bit integers. That means, we interpret the memory of the 32-bit float as 32-bit i [... cut]
commit 13bd5841 (2024-07-17): Fixes float encoding/decoding for both big and little endian (fixes #665, #694)

* Removes useless memcpy calls and no longer used swap32 and swap16 macros.
* Updated float encoding functions to consider endianess of 16-bit registers (fixes #665)
* Deprecated modbus_get_float() and modbus_set_float() are implemented using cdab variants.
commit 8e769e7a (2024-07-17): Revert "Fixes float encoding/decoding for both big and little endian (fixes #665, #694)"

This reverts commit 13bd58416b9e883fbec6b92879476b43e7902ccf.