stephane/libmodbus #674: src\win32\README.md instructions not working in Windows 10 with Visual Studio 2019

LamerTex, 2022-12-23:
## libmodbus version

  Tried both version 3.1.10 and latest commit in master branch

## OS and/or distribution

  Windows 10 Home 19045.2251 with Visual Studio 2019

## Environment

  both x86 and x64 have the same errors

## Description

After following the instructions in the README.md file the "Console App" project doesn't build

## Actual behavior if applicable

  ```Build started...
1>------ Build started: Project: CL3-E-1-0F_CommunicationTester, Configuration: Debug x64 ------
1>CL3-E-1-0F_CommunicationTester.cpp
1>modbus-data.c
1>D:\Working\LocalProjects\CL3E-CanOPEN_Motor\CL3-E-1-0F_CommunicationTester\CL3-E-1-0F_CommunicationTester\libmodbus\modbus-data.c(25,10): fatal error C1083: Cannot open include file: 'config.h': No such file or directory
1>modbus-rtu.c
1>D:\Working\LocalProjects\CL3E-CanOPEN_Motor\CL3-E-1-0F_CommunicationTester\CL3-E-1-0F_CommunicationTester\libmodbus\modbus-private.h(20,10): fatal error C1083: Cannot open include file: 'config.h': No such file or directory
1>modbus-tcp.c
1>D:\Working\LocalProjects\CL3E-CanOPEN_Motor\CL3-E-1-0F_CommunicationTester\CL3-E-1-0F_CommunicationTester\libmodbus\modbus-private.h(20,10): fatal error C1083: Cannot open include file: 'config.h': No such file or directory
1>modbus.c
1>D:\Working\LocalProjects\CL3E-CanOPEN_Motor\CL3-E-1-0F_CommunicationTester\CL3-E-1-0F_CommunicationTester\libmodbus\modbus.c(21,10): fatal error C1083: Cannot open include file: 'config.h': No such file or directory
1>Generating Code...
1>Done building project "CL3-E-1-0F_CommunicationTester.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
```

These first errors can be fixed by changing 
`#include <config.h>` 
to 
`#include "config.h"`
in the 4 header files.

After fixing this the build have the new error
```
Build started...
1>------ Build started: Project: CL3-E-1-0F_CommunicationTester, Configuration: Debug x64 ------
1>modbus.rc(3): fatal error RC1015: cannot open include file 'config.h'.
1>
1>Done building project "CL3-E-1-0F_CommunicationTester.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
```
which can be solved by changing in the modbus.rc file:
```
#include "config.h"
#include "modbus-version.h"
```
to
```
#include "libmodbus/config.h"
#include "libmodbus/modbus-version.h"
```
suggesting there is probably something wrong with the settings of the project even if the step 11
"in the **Property Pages** of the project *Configuration Properties ->
    Resources -> Additional Include Directories*, add `libmodbus` folder." should have fixed this.
Finally even specifying these header file the build fail with:
```
Build started...
1>------ Build started: Project: CL3-E-1-0F_CommunicationTester, Configuration: Debug x64 ------
1>
1>modbus.rc(8): error RC2135: file not found: LANGUAGE
1>
1>
1>modbus.rc(9): error RC2135: file not found: 1
1>
1>
1>modbus.rc(10): error RC2135: file not found: 3
1>
1>Done building project "CL3-E-1-0F_CommunicationTester.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
```

And from here I don't know how to go ahead to make it build correctly.

On the same computer using the [sln file made in this PR](https://github.com/stephane/libmodbus/pull/512) the library by itself compile correctly. So these instructions should be fixed or updated.

## Expected behavior or suggestion

  The created "Console App" should compile

## Steps to reproduce the behavior (commands or source code)

  Just follow the steps in src\win32\README.md
```
# Instructions to compile on Windows

## Create a new Visual Studio project with the library included

Download the latest version of libmodbus source code from Github
`https://github.com/stephane/libmodbus` -> Code -> Download ZIP.

Once the archive is decompressed, launch a Windows terminal (`cmd`) in
`src/win32` directory and run `cscript configure.js`.

1. copy the file `config.h` from `src/win32` to `src`.
2. create a new 'Console App' project under Visual Studio.
3. create a new directory called `libmodbus` inside your VS project (same level
   as the `.vcxproj` file).
4. copy all `*.c` and `*.h` from libmodbus `src` in the new `libmodbus` folder
   of your VS project.
5. copy `modbus.rc` in your VS project (same level as the `.vcxproj` file).
6. drag and drop `libmodbus/*.c` files (4) in *Solution Explorer -> Source Files*.
7. drag and drop `libmodbus/*.h` files (8) in *Solution Explorer -> Header Files*.
8. drag and drop `modbus.rc` file in *Solution Explorer -> Resource Files*.
9. check path is `#include "modbus-version.h"` in `modbus.rc`.
10. in the **Property Pages** of the project *Configuration Properties -> C/C++
    -> General -> Additional Include Directories*, add `libmodbus` folder.
11. in the **Property Pages** of the project *Configuration Properties ->
    Resources -> Additional Include Directories*, add `libmodbus` folder.
12. in the **Property Pages** of the project *Configuration Properties -> Linker
    -> Input*, define `ws2_32.lib`.
13. if required, add `_CRT_SECURE_NO_WARNINGS` to *C/C++ -> Preprocessor ->
    Preprocessor Definitions*.
```

alongL, 2022-12-28:
Just double click the configure.js configure.h  and some files will be generated.  
Open modbus-9.sln with  visual studio . 
you can build modbus.lib and modbus.dll.

LamerTex, 2022-12-28:
> On the same computer using the [sln file made in this PR](https://github.com/stephane/libmodbus/pull/512) the library by itself compile correctly. So these instructions should be fixed or updated.
Yes, I was able to build it in some other way as I've described in the "Expected behavior or suggestion" section, but only after losing a lot of time trying to follow the official steps described in src\win32\README.md

If the "official" steps for compiling the library in windows don't work they should be updated, even with just the steps you wrote here (maybe with a little more details). 
In my case, the "modbus-9.sln" was not working with VisualStudio 2019 and I had to use the changes made in PR #512 (which is still open) to make it build correctly.