stephane/libmodbus #675: configure.js generate version not correct

alongL, 2022-12-28:
Now the version of libmodbus is 3.1.10, but this script generate version  3.1.1  
Maybe verMicro = s.substr(s.indexOf(",") + 3, 1);   should be modified.
 s.substr(s.indexOf(",") + 3, 2) is rude.
I'm not familar with vbs. Who can do this?

```
function readVersion() {
    var fso, cf, ln, s;
    fso = new ActiveXObject("Scripting.FileSystemObject");
    cf = fso.OpenTextFile(configFile, 1);
    while (cf.AtEndOfStream !== true) {
        ln = cf.ReadLine();
        s = new String(ln);
        if (s.search(/^m4_define\(\[libmodbus_version_major/) != -1) {
            verMajor = s.substr(s.indexOf(",") + 3, 1);
        } else if (s.search(/^m4_define\(\[libmodbus_version_minor/) != -1) {
            verMinor = s.substr(s.indexOf(",") + 3, 1);
        } else if (s.search(/^m4_define\(\[libmodbus_version_micro/) != -1) {
            verMicro = s.substr(s.indexOf(",") + 3, 1);   //this line 
        }
    }
    cf.Close();
}
```

yauyimsing, 2023-06-09:
replace line 63 with follow: 

        } else if (s.search(/^m4_define\(\[libmodbus_version_micro/) != -1) {
			str_start_idx = s.indexOf(",") + 3;
			num_str_length = s.length - str_start_idx - 2;
            verMicro = s.substr(str_start_idx, num_str_length);
        }

stephane, 2024-07-17:
I don't have a win32 installation ready, in the meantime could you test https://github.com/stephane/libmodbus/commit/225fbbb175d3e795b6eacf12ab2a1705f3ea47e1 ?

closed 2024-07-17 by stephane