stephane/libmodbus #676: Broadcasting when RTU master...

dnssoftware, 2023-01-02:
When one is targeting slaves with MODBUS_BROADCAST_ADDRESS and say you issue modbus_write_registers,
the code after sending a frame still goes through 

if (rc > 0) {
        uint8_t rsp[MAX_MESSAGE_LENGTH];

        rc = _modbus_receive_msg(ctx, rsp, MSG_CONFIRMATION);
        if (rc == -1)
            return -1;

        rc = check_confirmation(ctx, req, rsp, rc);
    }

Shouldn't one bypass waiting for response as one isn't expected?

Regards

Dean

karlp, 2023-01-04:
in spec, yes, in reality, maybe.  I think this should be protected by a quirk mode.

bobemoe, 2023-07-30:
In my reality, definitely!

See https://github.com/arduino-libraries/ArduinoModbus/issues/117 where this lib is used on a microcontroller and causing an undesired delay. I have added a PR to that repo but its actually on the exact libmodbus code shown above.

bobemoe, 2023-07-30:
This is actually dupe of #379 which has a PR #467 to fix this!