stephane/libmodbus #686: libmodbus close() crash

znstj, 2023-02-27:
## libmodbus version

3.1.10

## OS and/or distribution

  Windows 7

## Environment

 Win7 32/64
 MinGW 32/64

## Description

Modbus RTU is frequently opened and closed,once every 15 seconds.
After a long run,(about 1~7days). modbus_read_registers() can collect data is correct, but modbus_close() is closed abnormally, causing the program to crash and occupy serial port resources, turning it into a zombie process.
        
```
        ctx = modbus_new_rtu(PortName.c_str(), BaudRate, serial_Parity, DataBits, StopBits);
        modbus_set_slave(ctx, ID);  
        rc = modbus_connect(ctx);
        if (rc == `-1)`
        {
            modbus_free(ctx);
        }
        else
        {
            rc = modbus_read_registers(ctx,0x0000,Read_Data_Len,&Data.data_buff[0]);
            if (rc != -1)
            {              
               ------
            }
            else
            {
              -----
            }    
            modbus_close(ctx);
            modbus_free(ctx);    
        }
```

karlp, 2023-02-27:
Do you have any more information?  You're talking about 1 in 10-30k times, so ~no-one is going to just run this again to try.  What does modbus_close print?  You're on windows, so you should be getting these at least.
https://github.com/stephane/libmodbus/blob/master/src/modbus-rtu.c#L1101-L1110

znstj, 2023-02-28:
> Do you have any more information? You're talking about 1 in 10-30k times, so ~no-one is going to just run this again to try. What does modbus_close print? You're on windows, so you should be getting these at least. https://github.com/stephane/libmodbus/blob/master/src/modbus-rtu.c#L1101-L1110

I can't capture this output error from the terminal, because this system is very complicated and the output information is too much.
May be add modbus_flush？

```
modbus_flush(ctx);
modbus_close(ctx);
modbus_free(ctx); 
```
But in modbus_flush，https://github.com/stephane/libmodbus/blob/master/src/modbus-rtu.c#L1121-L1130
Only `PurgeComm(ctx_rtu->w_ser.fd, PURGE_RXCLEAR)`
Perhaps this code is safer:
`PurgeComm(ctx_rtu->w_ser.fd, PURGE_RXCLEAR|PURGE_RXABORT);`

znstj, 2023-03-14:
OK,This is not a problem with libmodbus.
It may be caused by hardware or system!

closed 2023-03-14 by znstj