stephane/libmodbus #688: modbus_write_registers return -1, reason connection timeout

yanzhang920817, 2023-03-08:
use version 3.1.6 on linux, modbus_write_registers(ctx[index], addr, count, buf) return -1, modbus_strerror(errno) show connection timed out.

`int addr = 259;
int count = 1;
uint16_t buf[10]=[1000]`

The connection settings are as follows

`       modbus_set_byte_timeout(ctx[index], 0, 200 * 1000);
        modbus_set_indication_timeout(ctx[index], 0, 100 * 1000);
        modbus_set_response_timeout(ctx[index], 0, 200 * 1000);
        if (-1 == modbus_connect(ctx[index])) {
            SPDLOG_ERR(channel_http, "[ModbusCreate] connect {}:{} faild reason:{}", ip, port, modbus_strerror(errno));
            modbus_free(ctx[index]);
            ctx[index] = NULL;
            ret = "-1,{},";
            return;
        }
        int socket = modbus_get_socket(ctx[index]);
        int keepAlive = 1;       
        int keepIdle = 1;     
        int keepInterval = 1;    
        int keepCount = 3;    

        setsockopt(socket, SOL_SOCKET, SO_KEEPALIVE, (void*)&keepAlive, sizeof(keepAlive));
        setsockopt(socket, SOL_TCP, TCP_KEEPIDLE, (void*)&keepIdle, sizeof(keepIdle));
        setsockopt(socket, SOL_TCP, TCP_KEEPINTVL, (void*)&keepInterval, sizeof(keepInterval));
        setsockopt(socket, SOL_TCP, TCP_KEEPCNT, (void*)&keepCount, sizeof(keepCount));`

yanzhang920817, 2023-03-08:
if using modbus_set_byte_timeout, the modbus_set_response_timeout is not working?

yanzhang920817, 2023-03-08:
Maybe I should use modbus_ set_ indication_ Timeout (ctx [index], 0, 0), let the server wait all the time, so that there will be no connection timeout problem?

yanzhang920817, 2023-03-08:
Maybe I should use modbus_ set_ indication_ timeout (ctx [index], 0, 0), let the server wait all the time, so that there will be no connection timeout problem?

Reivax47, 2023-06-28:
Hi !
I'm having the same issue. Did you manage to find out ?

stephane, 2024-10-22:
Maybe your TCP connection is unstable...
In your example, you play with the socket so not very helpful to analyse the issue.

closed 2024-10-22 by stephane