stephane/libmodbus #706: [Segmentation fault] in libmodbus.so

albertZhuo420, 2023-07-06:
Please read the following carefully before submitting this new issue.

- Please ensure, that you are really reporting a bug. When in doubt, post a
  message on <https://groups.google.com/forum/#!forum/libmodbus> or send an
  email to libmodbus@googlegroups.com

- Please do not open issues to ask questions about using libmodbus. Use the
  mailing list for this as there are many more people reading that list, who
  could help you.

- When using libmodbus from a distribution (Debian, Fedora...), please report
  the bug first in the bug tracker of the distribution. The reason for doing so
  is that the package maintainer should have a chance to look at the issue first
  as it might be a packaging error. If/when the package maintainer comes to the
  conclusion that is really an upstream bug, then he/she will usually report it
  here by himself/herself. This is because he/she is interested in staying in
  the notification chain to decide about a backport as soon as a bugfix is
  available. Otherwise you (distribution user) will be asked to do so
  explicitly.

When you get here and you are still convinced that you want to report a bug:

- *Use a clear and descriptive title* for the issue to identify

- *Which version of libmodbus are you using?* you can obtain this information
   from your package manager or by running `pkg-config --modversion libmodbus`.
   You can provide the sha1 of the commit if you have fetched the code with `git`.

- *Which operating system are you using?*

- *Describe the exact steps which reproduce the problem* in as many details as
   possible. For example, the software/equipment which runs the Modbus server, how
   the clients are connected (TCP, RTU, ASCII) and the source code you are using.

- *Enable the debug mode*, libmodbus provides a function to display the content
   of the Modbus messages and it's very convenient to analyze issues
   (<http://libmodbus.org/docs/modbus_set_debug/>).

Good bug reports provide right and quick fixes!

Finally, thank you very much for using libmodbus and taking the time to
file a good bug report. Doing so signals your respect for the developers.

The following template helps you to address the points above. Please delete
everything up to and including the following line which starts with ---.

---

## libmodbus version

  <libmodbus3.1.6>

## OS and/or distribution

Centos7.9
$ cat /etc/redhat-release 
CentOS Linux release 7.9.2009 (Core)

## Environment

$ uname -a
Linux localhost.localdomain 3.10.0-1160.el7.x86_64 #1 SMP Mon Oct 19 16:18:59 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

## Description

$ dmesg
python3[14160]: segfault at 41 ip 00007f522acccd5a sp 00007ffedfce25e0 error 4 in libmodbus.so.5[7f522acca000+a000]

I use addr2line to check where the code which cause the segfault. I find that :

in file src/modbus.c, 516 line: [ const int offset = ctx->backend->header_length; ] in func: [static int check_confirmation()];

Note:  The error is not reported at the beginning, but only after running for a while, about 3 hours.
 
## Actual behavior if applicable

  <...>

## Expected behavior or suggestion

  <...>

## Steps to reproduce the behavior (commands or source code)

  <...>

## libmodbus output with debug mode enabled

  <...>

mhei, 2023-07-06:
Your issue would be more readable when you follow the instructions in the template:

> ... Please delete everything up to and including the following line which starts with...

So please edit it.

albertZhuo420, 2023-07-08:
Sorry, this is not your problem, it's my problem. I have already found the bug.

closed 2023-07-08 by albertZhuo420