stephane/libmodbus #722: Connection not bindig

fenilGhoghari, 2023-11-16:
Hello @stephane ,

I have used your library in QT framework and i am trying to make one QT application but when I first time launch my application at a time i am able to make connection but when i dis-connect with current client and using same IP and PORT at a time my server is not closed and try to connect with my client but i am not able to connect with my second client can you help me to close the Binding hear.

I am sure here port is not closed completely and this make issue to new connection. I didn't modify your disconnect function and i am using as it is you have written.

Environment I used:
System : Windows 10 and Debian
QT version: 5.15.0

How I tested?
-> First launch application using QT Creature
-> Launch Server
-> Connect with client ( Success)
-> Client Disconnect ( Success)
-> Relaunch Client ( Same IP and Port) **( Fail)**

Expected Outcome:
-> Server Able to connect with Client

I hope you will help me out to solve this issue I will be grateful to you if you help me here!!!

**My Code for Server**
```
if (strippedIP == "")
        {
            mainWin->showUpInfoBar(tr("Connection failed\nWrong IP Address."), MyInfoBar::Error);
            QLOG_ERROR() << "Connection failed. Blank IP Address";
            return;
        }
        else
        {
            m_modbus = modbus_new_tcp(strippedIP.toLatin1().constData(), port);
            mainWin->hideInfoBar();
            QLOG_INFO() << "Connecting to IP : " << ip << ":" << port;
        }
//        m_timeOut = timeOut;

        mb_mapping = modbus_mapping_new(MODBUS_MAX_READ_BITS, 0, MODBUS_MAX_READ_REGISTERS, 0);
           if (mb_mapping == NULL)
           {
               fprintf(stderr, "Failed to allocate the mapping: %s\n", modbus_strerror(errno));
               modbusDisConnect();
               return;
           }
           else
           {
               QLOG_INFO() << "Mapping Success";
           }
    server_socket = modbus_tcp_listen(m_modbus, NB_CONNECTION);

    if (server_socket == -1)
        {
            fprintf(stderr, "Unable to listen TCP connection\n");
            modbusDisConnect();
            return;
        }
        else
        {
            QLOG_INFO() << "Listening for connections...";
        }

    if(modbus_tcp_accept(m_modbus, &server_socket)<0){
        fprintf(stderr, "Failed to accept connection: %s\n", modbus_strerror(errno));
        m_connected = false;
    }else {
        // Connection established, start the data sending timer
        m_connected = true;
        data_sendTimer->start(1000);
```
**Disconnect Function**
```
void ModbusAdapter::modbusDisConnect()
{
    //Modbus disconnect

    QLOG_INFO()<<  "Modbus disconnected";

    if(m_modbus) {
        QLOG_INFO()<<  "Modbus disconnected process";
        modbus_mapping_free(mb_mapping);
        modbus_close(m_modbus);
        modbus_free(m_modbus);
        free(m_modbus->backend_data);
        m_modbus = NULL;
    }

    m_connected = false;

    m_ModBusMode = EUtils::None;

}
```
Thanks & Regards,
Fenil

ladmanj, 2023-11-20:
Hello @Fenil, @Stephane and all interested

That's the same behavior I reported few years ago.
Also Qt application, also copied the recommended opening and closing 
sequences from libmodbus docs.

Maybe we can focus on properly flushing all the data before closing the 
port. My micro-controller devices using code modified from libmodbus are 
going crazy when incomplete data frame is stuck in buffers.

The only one different correlation I can see is that in both cases, 
libmodbus code is called from c++ project.

Have a nice day

Jakub Ladman

stephane, 2024-10-22:
@ladmanj could you confirm the issue is still present (docs and code). I don't understand why calling `modbus_close` from C++ code would be an issue. Do you a snippet C++ code to compile to observe the issue?

BTW don't `free(m_modbus->backend_data)` after a call to `modbus_free`.

https://github.com/stephane/libmodbus/blob/master/src/modbus-rtu.c#L1169
https://github.com/stephane/libmodbus/blob/master/src/modbus-tcp.c#L816

closed 2025-01-27 by stephane