stephane/libmodbus #723: CRC Check Error and Slave not the requested One, Ignored

suresh-dersec, 2023-12-02:
Please read the following carefully before submitting this new issue.

- Please ensure, that you are really reporting a bug. When in doubt, post a
  message on <https://groups.google.com/forum/#!forum/libmodbus> or send an
  email to libmodbus@googlegroups.com

- Please do not open issues to ask questions about using libmodbus. Use the
  mailing list for this as there are many more people reading that list, who
  could help you.

- When using libmodbus from a distribution (Debian, Fedora...), please report
  the bug first in the bug tracker of the distribution. The reason for doing so
  is that the package maintainer should have a chance to look at the issue first
  as it might be a packaging error. If/when the package maintainer comes to the
  conclusion that is really an upstream bug, then he/she will usually report it
  here by himself/herself. This is because he/she is interested in staying in
  the notification chain to decide about a backport as soon as a bugfix is
  available. Otherwise you (distribution user) will be asked to do so
  explicitly.

When you get here and you are still convinced that you want to report a bug:

- *Use a clear and descriptive title* for the issue to identify

- *Which version of libmodbus are you using?* you can obtain this information
   from your package manager or by running `pkg-config --modversion libmodbus`.
   You can provide the sha1 of the commit if you have fetched the code with `git`.

- *Which operating system are you using?*

- *Describe the exact steps which reproduce the problem* in as many details as
   possible. For example, the software/equipment which runs the Modbus server, how
   the clients are connected (TCP, RTU, ASCII) and the source code you are using.

- *Enable the debug mode*, libmodbus provides a function to display the content
   of the Modbus messages and it's very convenient to analyze issues
   (<http://libmodbus.org/docs/modbus_set_debug/>).

Good bug reports provide right and quick fixes!

Finally, thank you very much for using libmodbus and taking the time to
file a good bug report. Doing so signals your respect for the developers.

The following template helps you to address the points above. Please delete
everything up to and including the following line which starts with ---.

---

## libmodbus version

  <libmodbus version or sha1 of the latest commit>

## OS and/or distribution

  <e.g. Windows, Linux... version>

## Environment

  <e.g. CPU architecture, 32 vs. 64 bit...>

## Description

  When I connect to the device, the first request is getting success and any subsequent query/request sent to my slave device from rtuclient (using libmodbus) resulting CRC check Error and Responding slave is not the requested one so Ignored.

## Actual behavior if applicable

  The same slave device is tested with modbus poll, modscan32 software and working fine.

## Expected behavior or suggestion

  <...>

## Steps to reproduce the behavior (commands or source code)

  <...>

## libmodbus output with debug mode enabled

  <...>
The client connection from 192.168.1.105 is accepted
Opening /dev/ttyUSB0 at 9600 bauds (N, 8, 1)
Waiting for an indication...
<00><01><00><00><00><06><01><03><02><49><00><03>
[01][03][02][49][00][03][D5][A5]
Waiting for a confirmation...
<01><03><06><00><F0><00><F0><15><9B><2E><68>

[00][01][00][00][00][09][01][03][06][00][F0][00][F0][15][9B]
Waiting for an indication...
<00><02><00><00><00><06><01><03><02><49><00><03>
[01][03][02][49][00][03][D5][A5]
Waiting for a confirmation...
<00><64><01><03><06>
ERROR CRC received 0x306 != CRC calculated 0x9AC0

[00][02][00][00][00][09][01][03][06][00][F0][00][F0][15][9B]
Waiting for an indication...
<00><03><00><00><00><06><01><03><00><01><00><03>
[01][03][00][01][00][03][54][0B]
Waiting for a confirmation...
<00><F0><00><F0><15>
ERROR CRC received 0xF015 != CRC calculated 0x35C0

[00][03][00][00][00][09][01][03][06][00][F0][00][F0][15][9B]
Waiting for an indication...
<00><04><00><00><00><06><01><03><00><01><00><03>
[01][03][00][01][00][03][54][0B]
Waiting for a confirmation...
<9B><2E><68><00><64>
Request for slave 155 ignored (not 1)
The responding slave 155 isn't the requested slave 1
[00][04][00][00][00][09][01][03][06][00][F0][00][F0][15][9B]
Waiting for an indication...
<00><05><00><00><00><06><01><03><00><01><00><03>
[01][03][00][01][00][03][54][0B]
Waiting for a confirmation...
<01><03><06><00><01><01><04><32><32><C8><3D>
[00][05][00][00][00][09][01][03][06][00][01][01][04][32][32]
Waiting for an indication...

robinmholt-daikin, 2025-07-03:
@stephane , we are running into this in our development environment and I expect we will run into it in production soon.

In our testing, we have libmodbus 3.1.6 and 3.1.10.  Our lead developer is working using 3.1.10, but that shouldn't make too much difference.

In their testing, they have a client configured to talk with 2 servers, but only one of the servers started.  Periodically, the client is sending a request to the other server and does not get a response (times out).  Once our server sees that request, it goes into the waiting for response mode, but does not get one.

I think there are two separate problems, but would like to discuss them more fully before heading for a solution.

1) When the second server is waiting for the response, it is not also checking to make sure the address for its next packet matches the address from the request.  I think that solves our immediate problem.

2) When we get into a framing error situation, I think there needs to be much better logic for regaining the framing.  I really wish modbus had some sort of frame marker, but... 1970's right!

For our proposed solution, I am thinking we need a state machine whose state indications are housed in the context.  As we are hitting a request for a server which is not ours, we record the server's id and go into the waiting for response mode.  If the server id for the next frame does not match, then we need to switch back to the waiting for request mode instead of waiting for response mode and expect bytes of the appropriate size.

rmcgre-daikin, 2025-07-03:
This only fixes one case. Suppose a client issues retries after it does not hear a response to its first request. Then, the next message seen by the second server will match the address from the request, but it is still not a response (it is just a duplication of the request)

https://www.modbus.org/docs/Modbus_over_serial_line_V1_02.pdf

One of the ways to get around this problem in principle would be to detect when a frame starts in a "robust" way - which according to the RTU specification is to wait for a 3.5-character-long gap on the wire. But this is probably unpracticable with delays from device drivers / kernel, etc.

This problem may require some more robust parsing to estimate when a frame is truly beginning.