stephane/libmodbus #727: incorrect type to store socket on Windows

Swiftkill, 2023-12-14:
modbus-private.h:100 int s;

SOCKET on Windows is not int, it's a larger type - int64_t*.

stephane, 2024-10-22:
Do you mean `int` are not set according to the underlying architecture on Windows (64 bits)?!

Swiftkill, 2024-12-21:
Not on all compilers, no, not in all compile modes either, and ISO standard doesn't prescribe it. You got no guarantee that happen. Tha's more typical for embedded or microcontroller stuff. Nether  x86-64 gcc nor MSVC does that (the latter is where I spotted the problem).  The headers Winsock.h and  Winsock2.h (POSIX-esque and async API) are ones where that type is defined , same headers are used both for x32 and x64:
```
typedef UINT_PTR        SOCKET;
```
basetsd.h:
```
#if defined(_WIN64)
    typedef __int64 INT_PTR, *PINT_PTR;
#else
    typedef _W64 int INT_PTR, *PINT_PTR;
#endif

```
```
#if !defined(_W64)
#if !defined(__midl) && (defined(_X86_) || defined(_M_IX86) || defined(_ARM_) || defined(_M_ARM)) && _MSC_VER >= 1300
#define _W64 __w64
#else
#define _W64
#endif
#endif
```
So on Windows you're safe to use `uintptr_t` for a cast SOCKET to. Random tests may show that everything is ok because descripor pointer value often stays low.  Infamouse Windows HANDLE is also a pointer to void now. it _used_ to be int, just as SOCKET in times of of Win98\Win2000, so you can find a legacy code that relies on it.