stephane/libmodbus #728: [Project] CLA processing is laggy (want to upstream support for Modbus over USB)

jimklimov, 2023-12-15:
Hello, got a question regarding the project rituals more than libmodbus itself at the moment: a NUT contributor implemented a patch for USB RTU for libmodbus, tested with a number of UPS devices. There is an intention to upstream this change (to avoid users having to custom-build libmodbus to get the connectivity with their UPSes), the author says he posted the CLA form some weeks ago... and is still waiting to be green-lit to propose pull requests. Can this be somehow faster? :)

References:

* https://github.com/networkupstools/nut/pull/2063
* https://github.com/EchterAgo/libmodbus/commits/rtu_usb

jimklimov, 2024-05-06:
Gentle bump :)

stephane, 2024-07-16:
I've just uploaded all the signed CLA to the bot.

closed 2024-07-16 by stephane

jimklimov, 2024-07-16:
Sorry @karlp the quick reply fell through the cracks; and thanks @stephane - this issue bubbled up in my mailbox again :)

As far as I know, there was no PR to the project posted yet regarding the `rtu_usb` feature, as the original author's (@EchterAgo) understanding was that a CLA must be in place before proposing code to this repository. BTW I do not see that nickname in https://github.com/networkupstools/libmodbus/commit/1c1ccc34dedb98f6b2c302a029df3a17555a44cd although per https://github.com/networkupstools/nut/pull/2063#issuecomment-1773569543 and https://github.com/networkupstools/nut/pull/2063#issuecomment-1848974437 the request was somehow posted :\

All that said, by now the code in the branch can be deemed fairly well tested in the field, several NUT users discussed building with it and eventually reported success, iterated on NUT-side improvements, etc. A slightly newer replica for NUT users who want to use their APC UPSes per https://github.com/networkupstools/nut/wiki/APC-UPS-with-Modbus-protocol instructions is maintained at https://github.com/networkupstools/libmodbus/tree/rtu_usb (same as the @EchterAgo's code, plus a README update).

jimklimov, 2024-07-16:
On a side note, per https://github.com/networkupstools/nut/pull/1671 and https://github.com/stephane/libmodbus/issues/659 here, there were other developments for NUT drivers that needed more libmodbus capabilities than currently provided by the project's main codebase.

If this recent activity is a sign of rekindled interest to the library from its maintainers, it would be great to also have "modbus-ASCII" and "read device identification" points solved in the core and not in a mish-mash of forks and cherry-picks :)

EchterAgo, 2025-01-16:
Sorry, I hadn't submitted my CLA request because I was not comfortable sharing my personal information without knowing how it is used. I sent a question regarding clarification by email but never got a response. I have now submitted the CLA with my companies address instead.

jimklimov, 2026-04-14:
For a bit of a heads-up, there is a recent burst of activity on the fork, with some fixes to the NUT driver and some to the library; hope to propose the upstreaming PR after that is well-tested. Kudos to @EchterAgo 
* https://github.com/networkupstools/libmodbus/pull/11