stephane/libmodbus #733: Missing documentation of `errno` value corresponding to Modbus exception codes

kyllingstad, 2024-01-05:
## libmodbus version

e6b71a88866b86c1cb66c2dafa5a08e6b87ad596 (which is the tip of `master` at the time of writing)

## Description

The libmodbus documentation does not specify how one can obtain information about Modbus exception codes returned by a server upon a failed read or write operation.

## Actual behavior if applicable

By reading the source code, I've discovered that libmodbus *does* in fact set `errno` to codes corresponding to the Modbus exception codes, such as `EMBXILFUN` etc. However, as far as I can tell, this is not mentioned anywhere in the reference documentation. For example, the reference page for `modbus_read_registers()` lists `EMBMDATA` as the only error code (which is an internal libmodbus error and not a Modbus exception). 

## Expected behavior or suggestion

Ideally, each function reference page should list all possible `errno` values, or at least the non-standard ones.

At minimum, the [Error handling](https://github.com/stephane/libmodbus/blob/master/docs/index.md) section of the front page should describe in general terms how Modbus exception codes are handled, what the corresponding constants are called, and which functions may return them.

kyllingstad, 2024-01-05:
I'd be happy to draft the minimal description I suggested at the end and submit a PR if this is desirable.

stephane, 2024-07-17:
Yes please!

PR #758 (2024-10-22): Document macros for error codes corresponding to Modbus exceptions
This fixes #733.
closed 2024-10-22 by stephane