stephane/libmodbus #751: Baud rate higher than 38400 causes Invalid Argument

Kektis, 2024-06-07:
Hello, I have a problem with my libmodbus modbus RTU server. I have a simple code for modbus RTU server. I use it in Embox, configurable RTOS and install it on stm32f746g-discovery. Developer for that RTOS says this might be problem with library, but I couldn't figure out what this is and can't find any information on the internet. 

Here's my code:

```
#include <stdio.h>
#include <errno.h>
#include <modbus.h>
#include <modbus-rtu.h>
#include <string.h>
#include <signal.h>
#include <stdlib.h>

const int REMOTE_ID = 10;
modbus_t *ctx = NULL;

void signal_handler(int signum) {
if (ctx) {
modbus_close(ctx);
modbus_free(ctx);
}
exit(signum);
}

void update_registers(modbus_mapping_t *mb_mapping) {
mb_mapping->tab_registers[0] = 1;
mb_mapping->tab_registers[1] = 0;
}

int main() {
uint8_t *request;
modbus_mapping_t *mb_mapping;
int rc;

signal(SIGINT, signal_handler);
signal(SIGTERM, signal_handler);

ctx = modbus_new_rtu("/dev/ttyS0", 115200, 'N', 8, 1);
if (ctx == NULL) {
fprintf(stderr, "Unable to create the libmodbus context\n");
return -1;
}

modbus_set_debug(ctx, TRUE);

if (modbus_set_slave(ctx, REMOTE_ID) == -1) {
fprintf(stderr, "Failed to set slave ID: %s\n", modbus_strerror(errno));
modbus_free(ctx);
return -1;
}

if (modbus_connect(ctx) == -1) {
fprintf(stderr, "Connection failed: %s\n", modbus_strerror(errno));
modbus_free(ctx);
return -1;
}

mb_mapping = modbus_mapping_new(MODBUS_MAX_READ_BITS, 0, MODBUS_MAX_READ_REGISTERS, 0);
if (mb_mapping == NULL) {
fprintf(stderr, "Cannot allocate mb_mapping: %s\n", modbus_strerror(errno));
modbus_close(ctx);
modbus_free(ctx);
return -1;
}

update_registers(mb_mapping);

request = malloc(MODBUS_RTU_MAX_ADU_LENGTH);
if (request == NULL) {
fprintf(stderr, "Unable to allocate request buffer\n");
modbus_mapping_free(mb_mapping);
modbus_close(ctx);
modbus_free(ctx);
return -1;
}

uint32_t response_timeout_sec = 1;
uint32_t response_timeout_usec = 0;

if (modbus_set_response_timeout(ctx, response_timeout_sec, response_timeout_usec) == -1) {
fprintf(stderr, "Unable to set response timeout: %s\n", modbus_strerror(errno));
free(request);
modbus_mapping_free(mb_mapping);
modbus_close(ctx);
modbus_free(ctx);
return -1;
}

for (;;) {
rc = modbus_receive(ctx, request);
if (rc > 0) {
printf("Received request: ");
for (int i = 0; i < rc; i++) {
printf("%02X ", request[i]);
}
printf("\n");

modbus_reply(ctx, request, rc, mb_mapping);
} else if (rc == -1) {
fprintf(stderr, "Error in modbus receive: %s\n", modbus_strerror(errno));
break;
}
}

free(request);
modbus_mapping_free(mb_mapping);
modbus_close(ctx);
modbus_free(ctx);

return 0;
}
```

Every time I set baud rate higher than 38400, I get 

`Opening /dev/ttyS0 at 115200 bauds (N, 8, 1) Connection failed: Invalid argument`

Every other baud rate works fine. What can cause it?

watsocd, 2024-06-07:
This is probably a physical problem, not a software issue.

 

The higher the baud rate, the more critical everything below becomes:

1.	Wiring techniques and quality: Network star points for example.
2.	Type of cable used
3.	Length of cable between master and all slaves
4.	Termination resistors placement and value
5.	Slave device capabilities
6.	Master Hardware used and signal strength

 

If your software works well at lower baud rates, there is a good chance there is nothing wrong with your code. However, watch out for certain timing parameters. For example, timing between requests. There needs to be time between master requests and slave responses to allow the network to settle electrically.

 

Chuck Watson

EnergyChaser, LLC

724-799-1529

karlp, 2024-06-07:
No, this is the underlying operating system saying that 38400 is not supported.  There's nothing libmodbus can do here, it asks to set a baudrate on a serial port, and the underlying OS is just saying "no. that's not a supported baud rate"

I can't see where in embox its failing, but it's definitely embox side

karlp, 2024-06-07:
probably right here? https://github.com/embox/embox/blob/06ac2cabf654316fbc794409aa4c440466135572/src/compat/posix/termios.c#L24-L33

Kektis, 2024-06-10:
> probably right here? https://github.com/embox/embox/blob/06ac2cabf654316fbc794409aa4c440466135572/src/compat/posix/termios.c#L24-L33

Thank you for your answer! I will ask the developer of OS to see that.

closed 2024-06-10 by Kektis