stephane/libmodbus #752: compile error

ruanyubiao, 2024-06-12:
modbus.c: In function ‘read_io_status’:
modbus.c:1202:9: error: ‘for’ loop initial declarations are only allowed in C99                                                                                                              or C11 mode
         for (unsigned int i = offset; i < offset_end; i++) {
         ^
modbus.c:1202:9: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 t                                                                                                             o compile your code
make[2]: *** [Makefile:484: modbus.lo] Error 1
make[1]: *** [Makefile:498: all-recursive] Error 1
make: *** [Makefile:384: all] Error 2

karlp, 2024-06-12:
time for you to update your compiler to one that defaults to c99.... :)

stephane, 2024-07-17:
The loop in `read_io_status` doesn't use loop initial declarations.
https://github.com/stephane/libmodbus/blob/136502367e077f840bbeb81488db82c55c99cca3/src/modbus.c#L1207

closed 2024-07-17 by stephane

stephane, 2024-07-17:
I'm agree with @karlp however that's the only place where a `loop initial declaration` is used so I edited this line.

commit 082d5323 (2024-07-17): Don't use loop initial declaration (closes #752)

C99 remains too modern :/