stephane/libmodbus #754: modbus_receive() function returns -1 on USB serial port (TTL level) devices

ZhiyuanYuanNJ, 2024-07-06:
- *Which operating system are you using?*
**Windows 10 x64**
- *Describe the exact steps which reproduce the problem* in as many details as
   possible. For example, the software/equipment which runs the Modbus server, how
   the clients are connected (TCP, RTU, ASCII) and the source code you are using.
**Modbus RTU**
I'm using modbus rtu, my code runs fine on a virtual serial port, but when I change the serial port to an actual TTL level USB serial port, the **modbus_receive()** function always returns -1.

- *Enable the debug mode*, libmodbus provides a function to display the content
   of the Modbus messages and it's very convenient to analyze issues
   (<http://libmodbus.org/docs/modbus_set_debug/>).

**The following is the debug report:**
.\RemoteGPIOControl.exe : ERROR timed out: select
所在位置 行:1 字符: 1
+ .\RemoteGPIOControl.exe COM37 COM70 2>out_1.txt
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (ERROR timed out: select:String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError

My Code Process:
```c++
// init 
modbus_new_rtu
modbus_mapping_new_start_address
modbus_set_slave
modbus_connect
do {
	ret = modbus_receive(MBInfo.mb, query);     //always return -1
	printf("ret == %d\n", ret); 
      } while (ret == 0);
.....
```

ZhiyuanYuanNJ, 2024-07-06:
It may be related to the manufacturer's driver used. I replaced the USB serial driver with VCP and the function It may be related to the manufacturer's driver used. I replaced the USB serial driver with VCP and the function did not return -1 did not return -1.

stephane, 2024-10-22:
The provided information aren't helpful to help you.

closed 2024-10-22 by stephane