stephane/libmodbus #769: How can I use libmodbus in Async

8188, 2024-09-18:
When I run the code below, **my client will only work after my server receives a query from another client**. It seems something might be blocking the process. 
However, when I split the code into two separate programs and run them simultaneously, everything works fine. 
I look forward to your help.

`class MyModbusServer {
	...
	void run()
	{
		int rc;
		while (1) {
			rc = modbus_receive(ctx.get(), query);
			if (rc == -1) {
				printf("Modbus client disconnected.\n");
				client_sock = modbus_tcp_accept(ctx.get(), &listen_sock);
				if (client_sock == -1) {
				}
			} else {
				printf("Received Modbus request (length=%d): ", rc);
				for (int i = 0; i < rc; i++)
					printf("%02X ", query[i]);
				printf("\n");
				rc = modbus_reply(ctx.get(), query, rc, mb_mapping.get());
				if (rc == -1) {
				}
			}
		}
	}
}

class MyModbusClient {
	...
	std::vector<uint16_t> read_registers(int start_registers, int nb_registers)
    {
        if (nb_registers <= 0) {
            return {};
        }

        std::vector<uint16_t> holding_registers(nb_registers);

        int blocks = nb_registers / MODBUS_MAX_READ_REGISTERS + (nb_registers % MODBUS_MAX_READ_REGISTERS != 0);

        try {
            for (int i = 0; i < blocks; ++i) {
                int addr = start_registers + i * MODBUS_MAX_READ_REGISTERS;
                int nb = (i == blocks - 1) ? nb_registers % MODBUS_MAX_READ_REGISTERS : MODBUS_MAX_READ_REGISTERS;
                int rc = modbus_read_registers(ctx.get(), addr, nb, &holding_registers[i * MODBUS_MAX_READ_REGISTERS]);
                if (rc == -1) {
                }
            }
        } catch (const std::exception& e) {
            std::cout << e.what() << '\n';
            holding_registers.clear();
            std::this_thread::sleep_for(std::chrono::seconds(5));
            connect(); // reconnect
        }
        return holding_registers;
    }
}

int main()
{
	auto modbusServer = std::make_unique<MyModbusServer>("xxx", 502, 1);

	auto serverFuture = std::async(std::launch::async, [&]() {
        std::cout << "Server run...";
        modbusServer.get()->run();
    });

    auto modbusCli = std::make_unique<MyModbusClient>("yyy", 502, 1);
    std::vector<uint16_t> registers;

    auto clientFuture = std::async(std::launch::async, [&]() {
        std::cout << "Client run...";
        while (true) {
            registers = modbusCli.get()->read_registers(0, 3);
			for (std::size_t i { 0 }; i < registers.size(); ++i) {
				std::cout << registers[i] << " ";
			}
			std::cout << '\n';
            std::this_thread::sleep_for(std::chrono::seconds(5));
        }
    });
	
	serverFuture.wait();
    clientFuture.wait();

    return 0;
}`

psychon, 2024-09-22:
Uhm... could you provide a reproducer of your problem? It is quite obvious that this code here never saw a compiler (and not just because of the missing `#include`s). 
```
error: expected ‘;’ after class definition
   24 | }
```
That one is quite easy to fix.
```
error: ‘ctx’ was not declared in this scope
    7 |             rc = modbus_receive(ctx.get(), query);
      |                                 ^~~
error: ‘query’ was not declared in this scope
    7 |             rc = modbus_receive(ctx.get(), query);

```
At this point I give up at guessing what your code is supposed to do.

8188, 2024-09-23:
@psychon I'm sorry. I have updated my codes.
My scenario involves using a Modbus client to receive information, perform calculations, modify mb_mapping, and then provide it to another client using a Modbus server.

psychon, 2024-09-23:
Thanks. That helps.

I actually had some trouble getting this to work and then I re-read what you wrote here:

> my client will only work after my server receives a query from another client. It seems something might be blocking the process.

The debugger says there is only a single thread. That thread is stuck in:
```
#2  0x000055555555e284 in MyModbusServer::init (this=0x5555555812b0) at foo.cpp:23
#3  0x000055555555e422 in MyModbusServer::MyModbusServer (this=0x5555555812b0, ip=0x555555564126 "127.0.0.1", port=1502, slave_id=1) at foo.cpp:41
#4  0x000055555555ff7e in std::make_unique<MyModbusServer, char const (&) [10], int, int> () at /usr/include/c++/14/bits/unique_ptr.h:1076
#5  0x0000555555558809 in main () at foo.cpp:116
```
The problem is that the constructor of `MyModbusServer` waits for an incoming connection (via `modbus_tcp_accept`) and blocks until then. Thus, the main thread is blocked in constructing `MyModbusServer` and not even the first call to `std::async` is reached.

8188, 2024-09-24:
@psychon Thanks a lot. I got it. I shouldn't put ```
listen_sock = modbus_tcp_listen(ctx.get(), 1);
        client_sock = modbus_tcp_accept(ctx.get(), &listen_sock);
``` in init().

closed 2024-09-24 by 8188