stephane/libmodbus #784: Building using CMAKE for C++ RAD Studio 12 ?

acronis2000, 2025-01-07:
Hello Stéphane !

Is there a way to add an assembly of your wonderful library using CMAKE for EMBARCADERO RAD Studio (C++) according to the instructions ?
[https://blogs.embarcadero.com/introducing-amazing-cmake-support-in-cbuilder-12-2](url)

It would be great if we realized this opportunity!

Thank you so much!

Duolabs, 2025-02-02:
Honestly? Don't to be rude but forget about RAD Studio and Idera products. Move your business to a much professional and stable IDE.
After paying 5k for a license I figured out that is not the right solution to develop your products unless you are hold tied on delphi only.

acronis2000, 2025-02-02:
I am quite satisfied with RAD Studio. 
Which IDE are you using that is more professional?

Duolabs, 2025-02-02:
Naturally, it depends on what you need to develop. I would start with Visual Studio, where the community version is free and allows you to avoid those annoying bugs from RAD Studio, where when you press CTRL on a function, it doesn't call it after 10 minutes of use.
A valid alternative is QT, although it has some limitations for commercial use, but in the end, it costs less than RAD Studio and is definitely the closest alternative.
For apps, there are countless options that allow deployment to iOS and Android without too much hassle.
Personally, I gladly gave up RAD Studio, also considering the fact that they are gradually abandoning C in favor of Delphi.
If you consider that you can't deploy C applications on macOS anymore from a certain version onward...
I don't develop software for release, I publish it only for internal use, so I base it on my own usage, which could be different from yours.

acronis2000, 2025-02-02:
Thank you so much for the very detailed answer!