stephane/libmodbus #797: Sugestion of a Custom Allocator

diplfranzhoepfinger, 2025-01-31:
in ESP32 we have different Memory: 

- malloc
- heap_caps_malloc(paSize, MALLOC_CAP_SPIRAM);

like in lua, and other Projects they use a custom allocator, where the App Programmer can decide how to allocate. 

maybe, if i have time, 
i make a suggestion.

WebDust21, 2025-02-03:
OR....you can just use the official ESP-Modbus library for the ESP32: https://github.com/espressif/esp-modbus

#dontreinventthewheel

diplfranzhoepfinger, 2025-02-03:
no. it has reasons that we are here at libmodbus. 

i know the esp-modus, and this one is not suitable for our needs. 
especially it is based on freemodbus not on libmodbus. 

we need libmodbus for 4diac integration, and here are more than just 1 person using libmodbus on esp32. 

moreover, if you program it on "C" the Interface of libmodbus is much easier to handle.