stephane/libmodbus #807: Floating point byte order does not round-trip

johannes1971, 2025-05-01:
Maybe I just completely misunderstand how this works, but is it the intention that modbus_set_float_<xxxx> does not round-trip with modbus_get_float_<xxxx>? I.e.
```
std::array<uint16_t, 2> arr;
modbus_set_float_abcd (24.2, arr.data ());
LOG (std::format ("abcd {}", modbus_get_float_abcd (arr.data ()))); // returns "-12.100244"

modbus_set_float_badc (24.2, arr.data ());
LOG (std::format ("badc {}", modbus_get_float_badc (arr.data ()))); // returns "-12.100244"

modbus_set_float_cdab (24.2, arr.data ());
LOG (std::format ("cdab {}", modbus_get_float_cdab (arr.data ()))); // returns "-12.100244"

modbus_set_float_dcba (24.2, arr.data ());
LOG (std::format ("dcba {}", modbus_get_float_dcba (arr.data ()))); // returns "-12.100244"
```
As it is, I have to apply a different byte order depending on whether I'm reading or writing, which feels rather counter-intuitive.

johannes1971, 2025-05-25:
Is there any news on the question I raised above? 

I'm after two pieces of information. The first is why round-tripping doesn't seem to work; this seems rather counter-intuitive to me, and I'd like to figure out if I'm doing something wrong. The second, perhaps more fundamental issue, is whether this library is still being maintained, or if the maintainers have given up and moved on.

tlbtlbtlb, 2025-06-25:
The code in modbus-data.c fails on some compilers such as clang-16 on Ubuntu on x86-64.

The problematic code is:
```
float modbus_get_float_abcd(const uint16_t *src)
{
...
  uint8_t a, b, c, d;
...

  i = (a << 24) | (b << 16) | (c << 8) | (d << 0);
```

A shift of a uint8 by more than 8 bits, as in (a<<24), is undefined behavior. You have to cast to uint32_t first.

A correct version is:
```
  i = ((uint32_t)a << 24) | ((uint32_t)b << 16) | ((uint32_t)c << 8) | ((uint32_t)d << 0);
```
and I can confirm that this roundtrips successfully on my machine.

I'm mystified as to why the test in test/unit-test-client.c doesn't catch this. Maybe the example number 123456.0 just happens to work.

psychon, 2025-07-12:
<details><summary>wise-cracking mode active, including a patch to make the test fail</summary>

> A shift of a uint8 by more than 8 bits, as in (a<<24), is undefined behavior.

Actually, no. Operator `<<` on a `uint8_t` and `24` (which is an `int`) first has to decide what type the result will be. For this, implicit integer promotion is applied. The common type between the two will be `int`, so `(a << 24)` will be interpreted as `((int)a << 24)`.

The undefined behavour happens when this overflows. If it does not overflow, there is no UB.

> I'm mystified as to why the test in test/unit-test-client.c doesn't catch this. Maybe the example number 123456.0 just happens to work.

123456.0 is 0x47f12000. This does not overflow an `int`. If you want an overflow, make this negative: -123456.0 is 0xc7f12000. (The highest bit in an IEEE 754 single precision number is the sign bit.)

The following patch:
```diff
diff --git a/tests/unit-test.h.in b/tests/unit-test.h.in
index 918cc43..f122dfc 100644
--- a/tests/unit-test.h.in
+++ b/tests/unit-test.h.in
@@ -60,9 +60,9 @@ const uint16_t UT_INPUT_REGISTERS_NB = 0x1;
 const uint16_t UT_INPUT_REGISTERS_TAB[] = { 0x000A };
 
 /*
- * This float value is 0x47F12000 (in big-endian format).
+ * This float value is -0xC7F12000 (in big-endian format).
  * In Little-endian(intel) format, it will be stored in memory as follows:
- * 0x00 0x20 0xF1 0x47
+ * 0x00 0x20 0xF1 0xC7
  *
  * You can check this with the following code:
 
@@ -74,7 +74,7 @@ const uint16_t UT_INPUT_REGISTERS_TAB[] = { 0x000A };
    }
    printf("\n");
  */
-const float UT_REAL = 123456.00;
+const float UT_REAL = -123456.00;
 
 /*
  * libmodbus will convert all words to processor-endianness as they come in and also converts them
```
causes:
```
TEST FLOATS
1/4 Set/get float ABCD: 
Line 326: assertion error for 'is_memory_equal(tab_rp_registers, UT_IREAL_ABCD, 4)': FAILED Set float ABCD
```
</details>

johannes1971, 2026-10-09:
Well, it appears to be fixed, so closing...

closed 2026-10-09 by johannes1971